Fornoth/spotify-connect-web #85: Volume controll is not working linear

tomekceszke, 2016-09-13:
Dear Maintainer,
first of all many thanks for that amazing tool!

I noticed that volume level that is passed from Spotify to Alsa thru spotify-connect-web is somehow converted. Unfortunately that conversion is not linear, but exponential (pic rel).
Is there a chance to have 1:1 conversion? I mean 50% in Spotify is equal 50% in alsa(mixer)?

My environment:
Raspberry Pi 2 
OS: OSMC (Debian based)
Spotify-connect-web release 0.0.3

[image]

tjerkdekuijper, 2016-09-15:
I can confirm this issue. I have got the same problem on my Raspberry Pi 3 with OSMC

flori4n, 2016-09-15:
Try to start spotify-connect-web.sh with --dbrange 25
this should help

awiouy, 2016-09-15:
@flori4n is there a systematic way to compute dbrange?

flori4n, 2016-09-15:
@awiouy Sorry not that I know. I'll try different --dbrange and see witch one suit best for me.

tomekceszke, 2016-09-15:
@flori4n many thanks for that hint - now volume control works as I want

Fornoth, 2016-10-02:
See #22, this is a hardware issue unfortunately. It's because devices report a larger volume range than they can actually handle

Fornoth, 2016-12-12:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-12-12 by Fornoth

bartberden, 2017-01-24:
It seems I'm having this issue as well.. Im using a IQaudio DAC, and my volume doesn't scale properly (below 50% i hear nothing, so my range is 50-100%). As I'm not that technical, what should I do to fix this?