Fornoth/spotify-connect-web #86: Cant adjust volume via Spotify

linuscool2, 2016-09-17:
Hello

I have a problem with my volume. 
I have installed Spotify-connect-web on my Raspberry Pi 2 with a USB-card and running the application with 
`$ ./spotify-connect-web --device 1 `

Now I cant change the volume. It worked for the first 5 minutes but now it dont.
I have restarted everything that has something to do with this but it wont work.

Help

Fornoth, 2016-10-02:
Try using the `--mixer` option. Grab the proper name using `amixer`

binderth, 2016-10-06:
Got the same issue. If using the USB card, amixer knows the device (and volume can be controlled) with name "speaker". Using that argument, spotify-comnect-web tells me, there's no device with that name.

blubhimself, 2016-10-06:
Try it with spotify-connect-web-chroot, then copy your asound.conf to spotify-connect-web-chroot/etc/ and start the server using --playback-device speaker.

binderth, 2016-10-07:
I already got the chroot-version. Now I double-checked and installed the 0.0.3-alpha as described in the Readme.md (./spotify-connect-web.sh install from https://github.com/Fornoth/spotify-connect-web/releases/download/0.0.3-alpha/spotify-connect-web.sh) - but I'll get the same errors:

`/root/spotify-connect-web.sh --name Music_WoZi --username xxx --password yyy --bitrate 320 --playback-device Speaker`
results to
`Loading Spotify library...`
`min_volume_range: 0.0`
`usage: main.py [-h] [--device DEVICE | --playback_device PLAYBACK_DEVICE]
               [--mixer_device_index MIXER_DEVICE_INDEX] [--mixer MIXER]
               [--dbrange DBRANGE] [--debug] [--key KEY] [--username USERNAME]
               [--password PASSWORD] [--name NAME] [--bitrate {90,160,320}]
               [--credentials CREDENTIALS]`
`main.py: error: unrecognized arguments: --playback-device Speaker`

how can I be sure, to have the newest 0.0.3 installed?

thanks,
Thomas.

blubhimself, 2016-10-07:
Try using _ insted of - in --playback_device.

"/root/spotify-connect-web.sh --name Music_WoZi --username xxx --password yyy --bitrate 320 --playback_device Speaker" should work for you.

binderth, 2016-10-07:
sorry. My bad! Got that, too!
Still some Troubles:
`root@MusicBox:~/spotify-connect-web-chroot# /root/spotify-connect-web.sh --name MusicBox_WoZi --username xxx --Password yyy --bitrate 320 --playback_device Speaker`
`Loading Spotify library...`
`min_volume_range: 0.0`
`Using libspotify_embedded version: master-v1.5.0-gfba69d71`
`SpInit: 0`
`playback_volume: 24903`
`corected_playback_volume: 37`
`public key: xxx`
`device id: 9eb01524-baef-437d-affa-0be476f425ff`
`remote name: sicBox_WoZi`
`account req: MIUM`
`device type: NOWN`
`kSpConnectionNotifyLoggedIn`
`TQ+GfTycS/jl75wMF11KnzHRy3MnpomkUEXMkQirCSqOwbVgIJ9D`
`kSpPlaybackNotifyPlay`
`playback_seek: 149206`
`kSpPlaybackNotifyTrackChanged`
`kSpPlaybackNotifyBecameActive`
`kSpPlaybackNotifyPlay`
`ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM Speaker`
`Unable to acquire device:  No such file or directory [speaker]`
`kSpPlaybackEventAudioFlush`

blubhimself, 2016-10-07:
Can you please post your asound.conf and the output of aplay -L ?

binderth, 2016-10-07:
```
root@MusicBox:~# /root/spotify-connect-web-chroot/usr/bin/aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #
```

alsa.conf is attached, only thing changed ideed  is
```
defaults.ctl.card 1
defaults.pcm.card 1
```
[alsa.conf.txt](https://github.com/Fornoth/spotify-connect-web/files/515487/alsa.conf.txt)

blubhimself, 2016-10-07:
Okay you need to create a file named "asound.conf" in your /etc folder which contents

pcm.softvol {
type softvol
slave {
pcm "default"
}
control {
name "Master"
card 1
}
}

Than copy that file to /root/spotify-connect-web-chroot/etc

Now start the Server:
/root/spotify-connect-web.sh --name MusicBox_WoZi --username xxx --Password yyy --bitrate 320 --playback_device softvol

binderth, 2016-10-07:
THANKS!!! That did it. I'm not so into ALSA, what does that "softvol"-entry do in short?

btw, I already got one asound.conf (first to entries I left in):
```
pcm.!default {
    type plug
    slave.pcm {
        type dmix
        ipc_key 1024
        slave {
            pcm "hw:1"
            rate 44100
        }
    }
}
ctl.!default {
    type hw
    card 1
}

pcm.softvol {
    type softvol
    slave {
        pcm "default"
    }
    control {
        name "Master"
        card 1
    }
}

```

blubhimself, 2016-10-07:
Great. The pcm.sofvol creates a new device named softvol (could be any other name)... the "type softvol" uses the ALSA Plugin softvol to add a new volume control to your hw card x  (in this case card 1) with the Mixer "Master" (can be seen in ALSA by typing "alsamixer" as root).

Fornoth, 2016-12-12:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-12-12 by Fornoth

binderth, 2016-12-12:
Everything works fine - after a bit of Trial and Error I also got my Pi to play mono (every room has only one Speaker).
```
pcm.softvol {
	// creates a device called softvol; type=softvol adds volume control to card 1 with Mixer "Master"
	type softvol
	slave {
		pcm "default"
	}
	control {
		name "Master"
		card 1
	}
}

pcm.spotify {
	// creates a device "spotify", which uses the above softvol-device - and as type=route it can downmix the channels to mono
	type route
	slave.pcm "softvol"

	ttable {
		# Copy both input channels to output channel 0 (Left).
		0.0 1
		1.0 1
		# Copy both input channels to output channel 1 (Right).
		0.1 1
		1.1 1
	}
}
```

TheBigFatTony, 2017-02-14:
Hello! I believe I have a similar problem and I'd be very glad if you could help me.
I have Spotify Connect 0.0.3-alpha installed on my RPi1B with a USB sound card. I can start Spotify Connect as follows, and it works, except that I can't change the volume:
`/home/pi/spotify-connect-web.sh --playback_device default:CARD=Device`
(Changing the volume on the Spotify Android app changes the volume of "PCM" of sound card 0.)
I'd like to apply the same trick as you guys did, but I don't understand the system sufficiently to do it myself. One difference could be that my Spotify Connect is installed in my home dir, and that I don't start it as root.

My /home/pi/.asoundrc:
```
ctl.!default {
    type hw
    card 1
}

pcm.softvol {
    type softvol
    slave {
        pcm "default"
    }
    control {
        name "Master"
        card 1
    }
}
```
My `aplay -L` output:
```
null
    Discard all samples (playback) or generate zero samples (capture)
softvol
default:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
default:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions
```
Running `/home/pi/spotify-connect-web.sh --playback_device softvol` runs, but it doesn't play sound through the USB sound card and changing the volume impacts "PCM" of sound card 0.

coda4k, 2017-02-15:
@TheBigFatTony , try to add "--mixer_device_index 1" to your options. I hope it helps!

TheBigFatTony, 2017-02-16:
@coda4k Thanks for the reply. It didn't work, I tried the following commands:
`./spotify-connect-web.sh --bitrate 320 --name RaspberryPi --playback_device default:CARD=Device --mixer_device_index 1`
and
`./spotify-connect-web.sh --bitrate 320 --name RaspberryPi --playback_device softvol --mixer_device_index 1`
and got the same error message both times:
```
Loading Spotify library...
Traceback (most recent call last):
  File "main.py", line 14, in <module>
    from connect import Connect
  File "/usr/src/app/connect.py", line 9, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/usr/src/app/console_callbacks.py", line 104, in <module>
    mixer = alsa.Mixer(args.mixer, **mixer_card_arg)
alsaaudio.ALSAAudioError: Unable to find mixer control PCM,0 [hw:1]
```