Fornoth/spotify-connect-web #87: Cannot find card 'default' (with chroot based on 0.0.3)

axelsimon, 2016-09-20:
Hi,
I updated my install today to use the latest chroot and I'm now running into an issue where the spotify-connect-web service runs fine, but as soon as I try to play something, it informs me that the default sound card can't be found:

> kSpPlaybackNotifyBecameActive
> kSpPlaybackNotifyPlay
> ALSA lib confmisc.c:768:(parse_card) cannot find card 'default'
> ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_card_driver returned error: No such device
> ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
> ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_concat returned error: No such device
> ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
> ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such device
> ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such device
> ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
> Unable to acquire device:  No such device [default:CARD=default]

At first I thought I might have messed up the config inside the chroot, as I'd messed about trying to get spotify-connect-web to play to a fifo file ALSA PCM device (attempting to use Snapcast). But I reinstalled a fresh chroot and the problem is still here.
If I start it with `./spotify-connect-web.sh  --bitrate=320 --name=test --playback_device default:CARD=ALSA` then it works fine.

Lastly, starting the old chroot with --device (as it was called then) works fine.

Any ideas?

twenta, 2016-09-20:
Did you follow this setup of the asound.conf?
https://github.com/badaix/snapcast/issues/45

axelsimon, 2016-09-21:
I did indeed, I created an .asoundrc file for my osmc user (I'm using OSMC, as you might have cunningly guessed from that) with exactly this:
```
osmc@capybara:~$ cat .asoundrc 
pcm.!snapcast {
type plug
slave {
pcm rate48000Hz # Direct default output to the below converter
}
}

pcm.rate48000Hz {
type rate
slave {
pcm writeFile # Direct to the plugin which will write to a file
format S16_LE
rate 48000
}
}

pcm.writeFile {
type file
slave.pcm null
file /tmp/snapfifo
format raw
}
```

I also tried with the same thing in `/etc/asound.conf`, it gives the same results, which is to be expected.

 `aplay -L` gives:
```
osmc@capybara:~$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
snapcast
rate48000Hz
writeFile
default:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
```
To test the spotify-connect-web issue, I tested without any asound.conf/.asoundrc. Still had the problem described in the title of this issue: “cannot find card 'default'”.
The same issue happens with device `snapcast`. It even happened with the `null` device when I tried last night but seems to work now.

Testing with a non-existant ALSA device ("zorg"):
```
kSpPlaybackNotifyPlay
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM zorg
Unable to acquire device:  No such file or directory [zorg]
```

Testing with the snapcast device:
```
kSpPlaybackNotifyPlay
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM snapcast
Unable to acquire device:  No such file or directory [snapcast]
```
Testing with the null device:
```
kSpPlaybackNotifyPlay
AlsaSink: device acquired
```
And for completeness, testing without giving a specificic playback device or giving `--playback_device default` (or `--playback_device=default`, same result):
```
kSpPlaybackNotifyPlay
ALSA lib confmisc.c:768:(parse_card) cannot find card 'default'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_card_driver returned error: No such device
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_concat returned error: No such device
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such device
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such device
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
Unable to acquire device:  No such device [default:CARD=default]
```
Lastly, it works fine when giving the full ALSA device:
```
kSpPlaybackNotifyPlay
AlsaSink: device acquired
```

twenta, 2016-09-21:
Try to add "-o snapcast" when you run spotify-connect-web.

coda4k, 2016-09-30:
spotify-connect-web chroot doesn't read asound.conf from your system's /etc directory. You will need to copy your /etc/asound.conf to your <spotify-connect-web-chroot>/etc/ directory.

I hope it helps!

Fornoth, 2016-10-02:
You can use `aplay -L` to figure out the name to pass to `--playback_device`.

axelsimon, 2016-10-04:
@coda4k Thanks, that sounds helpful, I'll give a try when I'm back near that install (it's not at my house).

@twenta @Fornoth Yes, I know, that's what I extensively tested and reported on in my previous post.
The whole problem is passing a device's name to spotify-connect-web with `./spotify-connect-web.sh --bitrate=320 --name=test --playback_device default` (default, snapcast) doesn't work.

It throws this error:

> ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such device
> ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
> Unable to acquire device: No such device [default:CARD=default]

So basically, it looks for the device  `default:CARD=default` instead of `default`.

singhprd, 2016-10-04:
Hi, Just thought I'd report back my progress with what I think is the same issue.
I have a raspberry pi 3 and want to output audio via the audio jack which is set to default in raspi-config.

I was getting this and no audio output:
`192.168.1.212 - - [2016-10-04 23:27:44] "GET /api/info/metadata HTTP/1.1" 200 596 0.006690
kSpPlaybackEventAudioFlush
kSpPlaybackNotifyPlay
ALSA lib conf.c:4687:(snd_config_expand) Unknown parameters CARD=default
ALSA lib pcm.c:2217:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
Unable to acquire device:  Invalid argument [default:CARD=default]`

I checked my aplay -L where the first entry was:
`aplay -L                              null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device`

I set my startup params to this which solved the issue and I now have audio output:
`./spotify-connect-web --playback_device sysdefault:CARD=ALSA`

Also thanks to everyone who is developing this! It was exactly what I was looking for on my pi!

Zeppp3, 2016-10-30:
Amen!

To get the sound working you need to put the full name `aplay -L` spits out. Like so:

--playback_device **sysdefault:CARD=ALSA**