Fornoth/spotify-connect-web #88: Shuffle not working as expected

sophof, 2016-09-24:
First of all, thanks for a great implementation already. I have it currently running on my Rpi1 but immediately ran into an issue.

Whenever I play my playlists using shuffle (which is always), it seems that on going to the next song, it reshuffles the entire playlist. Even though that seems truer to form, it's not what I want or what my android and desktop player does. There the playlist is shuffled once and then the queue is respected. In this way you can see the next upcoming song and add songs to be played next.

Not sure if this is actually a librespot issue, but I couldn't find any other mention of it. Is there any obvious way to fix this or adjust this to preference (in case this is intentional)?

Fornoth, 2016-10-02:
I have no control over how shuffle works :( this program currently uses spotify's proprietary `libspotify_embedded`, not  librespot
Also, queue'ing doesn't work with `libspotify_embedded` unfortunately

Fornoth, 2016-11-06:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-11-06 by Fornoth