Fornoth/spotify-connect-web #89: Question: use 7.1 USB Soundcard on pi for multiple instances?

binderth, 2016-09-28:
Hi there,
biiiiig shoutout for this great piece of Software! Thanks!
Now to my question: Would it be possible to use the channels of a 7.1 USB soundcard (it has raspbian Drivers) for multiple instances on a pi3?
What I would like to achieve is to have the each of the (single) loudspeakers in four rooms to function as different spotify connects. So it would be possible, my Kids listen each to their Music as my wife and me in different rooms.

Thanks,
Thomas.

Fornoth, 2016-10-02:
I'm not sure actually. Maybe you can set pulseaudio/alsa to have each channel as a separate virtual device.
This also looks useful http://askubuntu.com/a/440123

binderth, 2016-10-04:
Thanks! I'll have a look on this! Would spare me some money on dedicated PIs per room! *G*

closed 2016-12-12 by Fornoth