Fornoth/spotify-connect-web #91: how to setup for Spotify Family?

binderth, 2016-10-11:
I read the https://github.com/Fornoth/spotify-connect-web/issues/38 discussion, but please excuse me for asking, how to Setup all the rest of the config.
First of all, I'm using the chroot-Version and with some help here I got it setup using an external USB-soundcard on the pi. Now, the Goal is to Setup spotify-connect-web with our family logins. I'd like to have my Kids using it from their devices (with their family-logins) also. 
What I don't get is, how to configure spotify-connect-web with the zeroconf-login. I'm not familiar with Avahi, so perhaps can someone point me in the right direction?

Thanks!
Thomas.

karower, 2016-10-14:
I did the following:
`sudo apt-get install avahi-utils`
then editing a new script:
`nano /home/osmc/Skripte/spotify_multiuser.sh`
with this content:
`avahi-publish-service TestConnect _spotify-connect._tcp 4000 VERSION=1.0 CPath=/login/_zeroconf`
then:
`chmod u+x /home/osmc/Skripte/spotify_multiuser.sh`
and:
`sudo nano /etc/rc.local`
add before exit 0:
`/home/osmc/Skripte/spotify_multiuser.sh &`
and reboot

Unfortunately the first two letters of the name are missing within the spotify app.

binderth, 2016-10-14:
Hi, 

thanks for this, it'll start zeroconf with every start. Where can I configure the audio device with this? What's not clear to me, how I can tell the family Android devices, that there's a new Spotify Connect device...? Sorry, if I asked something obvious, but I don't get it right now! ;)

Thanks,
Thomas.

karower, 2016-10-14:
It will appear automatically in the app. For example I called it "Kueche" and it appears as "eche" in the app with another account.
Furthermore I didn't understand what else problems you have.

binderth, 2016-10-14:
Thanks for your help.
My problem is, I use an USB-soundcard, therefore I have to tell spotify-connect-web, which device to use (see https://github.com/Fornoth/spotify-connect-web/issues/86).

When I start zeroconf, my main account (the one I created the key for), sees the device and can use it - but there's no Sound, because the USB-soundcard isn't used. The account for my son sees the device, but can't use it, there's a error message popping up.

That's why I asked where I can configure something... ;)

karower, 2016-10-14:
I guess I can't help you with this. I have to wait 1-2 minutes after changing accounts. Afterwards I can use spotify connect with my i2s DAC.

seniorgod, 2016-10-16:
Hi all,

i also start to use spotify and set up spotify connect and the anahi deamon. 
I wonder if it is really necessary to publish the service after every start of the system. Is there more "memory" at avahi?

pbawn, 2016-10-16:
Hi,
The avahi-publish-service command does just as its name says: it registers a service that will be published by the avahi daemon. It won't start the service or interfere in its configuration, it's just advertisement. If you can connect without error but the spotify-connect-web service doesn't output any sound, the problem is most probably that spotify-connect-web wasn't launched with working _--playback_device_ and _--mixer_ parameters in the first place. Finding out which ones give the best result seems to be a matter of trial and error.
Zeroconf works pretty well by me. You can switch accounts instantly but have to pause and resume playback in the Spotify client app after having connected in order to get the sound out.

Fornoth, 2016-11-07:
@binderth so sound output works before you change accounts? maybe try @pbawn 's above solution? 
> You can switch accounts instantly but have to pause and resume playback in the Spotify client app after having connected in order to get the sound out.

binderth, 2016-12-03:
Ok. I gave it another try.
1) reinstalled Rasbian Jessie lite
2) installed the 0.3.0 Alpha chroot Version, copied spotify-key
3) got it to work with `./spotify-connect-web.sh --username xxx--password yyy --playback_device hw:1` 

So at this stage I can use it with my main spotify user. (without volume control, but one step after another)

So now I'd like to use it throughout my family account, and I did the following ( https://github.com/Fornoth/spotify-connect-web/issues/91#issuecomment-253795631 )
1) `sudo apt-get install avahi-utils
`2) `nano /home/pi/spotify_multiuser.sh`
with this content:
`avahi-publish-service Wohnzimmer _spotify-connect._tcp 4000 VERSION=1.0 CPath=/login/_zeroconf`
3) `chmod u+x /home/osmc/Skripte/spotify_multiuser.sh`
4) start the Multiuser script `/home/pi/spotify_multiuser.sh`

After this, there's no "Wohnzimmer" in my App or Spotify Dekstop to choose from. 

so, two questions:
1) What did I do wrong?
2) How could I tell the avahi-publish-service to use `hw:1` as `--playback_device`?

Thanks for help!

binderth, 2016-12-03:
oh... NOW I get it! 
The `avahi-publish-service` runs in the background and I have to start `./spotify-connect-web.sh` parallel.
So, after done this, everything works just fine!

Is there a way to start `./spotify-connect-web.sh` with the system after a reboot? I did include the script in the rc.local as here described: https://github.com/Fornoth/spotify-connect-web/issues/91#issuecomment-253795631 , but there-#s some strange error in the console, that the mount to chroot didn't work out?

binderth, 2016-12-12:
If you're using one speaker in each room (as I do), perhaps my chroot-asound.conf helps you to play mono there.
```
pcm.softvol {
	// creates a device called softvol; type=softvol adds volume control to card 1 with Mixer "Master"
	type softvol
	slave {
		pcm "default"
	}
	control {
		name "Master"
		card 1
	}
}

pcm.spotify {
	// creates a device "spotify", which uses the above softvol-device - and as type=route it can downmix the channels to mono
	type route
	slave.pcm "softvol"

	ttable {
		# Copy both input channels to output channel 0 (Left).
		0.0 1
		1.0 1
		# Copy both input channels to output channel 1 (Right).
		0.1 1
		1.1 1
	}
}
```
you can use it like this: 
`./spotify-connect-web.sh --username xxx--password xxx --bitrate 320 --name roomname --playback_device spotify` and it will show up as spotify connect device in the App of each family-connected spotify user.

binderth, 2016-12-12:
and I also got the autostart-issue live - created a crontab "@reboot" like shown in https://github.com/Fornoth/spotify-connect-web/issues/49

closed 2016-12-12 by binderth

CarlosMC, 2016-12-13:
hi all, 

i'm using LibreElec on rpi3 and the avahi-services it's different: 

rpi:~ # avahi-
avahi-browse          avahi-browse-domains  avahi-daemon          avahi-publish         avahi-set-host-name

rpi:~ # avahi-publish --service TestConnect _spotify-connect._tcp 4000 VERSION=1.0 CPath=/login/_zeroconf 

So, the name changed and it's necessary so add the --service parameter

rpi:~ # avahi-publish -V 
avahi-publish 0.6.32

Kind Regards,

awiouy, 2016-12-13:
@CarlosMC with LibreELEC:
1. [avahi-publish only ships with the spotify-connect-web addon](https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/addons/service/spotify-connect-web/package.mk#L50-L51)
2. [the spotify-connect-web addon starts avahi-publish as you mention](https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/addons/service/spotify-connect-web/source/bin/spotify-connect-web.start#L103-L104)