Fornoth/spotify-connect-web #94: Is it possible to make device active from API?

cadavre, 2016-11-17:
Is it possible to make `spotify-connect-web` active from API/library? After I turn on `spotify-connect-web` it is visible in Spotify clients, but it's not active by default. Can I request `active` state somehow?

Zalastax, 2016-12-05:
It's not possible to do in the current version of spotify-connect-web. You could see what happens if you allow the web interface to play even though it's not active. You could also try to see if libspotify has any function that would allow you to activate.

Fornoth, 2016-12-12:
I think if you call the spotify-connect-web API like `/api/playback/play` (ex: `localhost:4000/api/playback/play) it should activate the device and start playback.
Otherwise you could use something like https://github.com/badfortrains/spotcontrol to make it the active device