Fornoth/spotify-connect-web #95: Ups! Some badness happened:  Data size must be a multiple of framesize

redvoodoochild, 2016-11-22:
Hi.
Excuse my poor english.
I've got the same problem of this issue :  #64 
I'm trying to run spotify-web-connect on Libreelec with my usb DAC.

```
LibreELEC:~/.kodi/addons/service.spotify-connect-web# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
default:CARD=DA300USB
    DA-300USB, USB Audio
    Default Audio Device
sysdefault:CARD=DA300USB
    DA-300USB, USB Audio
    Default Audio Device
front:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    Front speakers
surround21:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
```
```
LibreELEC:~/.kodi/addons/service.spotify-connect-web # ./spotify-connect-web -n Salon -o iec958:CARD=DA300U
SB,DEV=0
Loading Spotify library...
min_volume_range: 0.0
Using libspotify_embedded version: release-esdk-1.20.0-v1.20.0-g594175d4
SpInit: 0
playback_volume: 60947
corected_playback_volume: 92
public key: xxxxxxxxxxxxx
device id: 0970b14f-86f5-4f55-983e-3590f68d4ffa
remote name: Salon
account req: PREMIUM
device type: AUDIODONGLE
kSpConnectionNotifyLoggedIn
iXFWVkc42sVam1QsiOwJcpjBpgSjtoI4QvujQJQJng2kRmja9fTCS5d0ZTpje/FS
kSpPlaybackNotifyPause
kSpPlaybackNotifyShuffleEnabled
playback_seek: 56863
kSpPlaybackNotifyTrackChanged
kSpPlaybackNotifyBecameActive
playback_seek: 56863
kSpPlaybackEventAudioFlush
kSpPlaybackNotifyPlay
AlsaSink: device acquired
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
kSpPlaybackNotifyPause
AlsaSink: device released
^CkSpPlaybackEventAudioFlush
kSpConnectionNotifyLoggedOut

```
There's no sound.
My DAC works very well with mpd

Thnak you for your help.

cadavre, 2016-11-23:
I had similar issues when my PCM was set to `rate 48000` and config inside of `console_callbacks.py` to 44100. Take a look at this settings.

redvoodoochild, 2016-11-24:
Thank you for your help.
I haven't set any configuration on alsa.
With MPD my souncard works with a rate of 44100.
And I don't know how to see or set the rate of my PCM, on libreelec.

tommph, 2016-11-26:
I had this issue with an asynchronous dac that would only work with S32LE.

Try `nano /storage/.config/asound.conf` and insert the following contents.

```
pcm.plug {
        type plug
        slave.pcm "hw:1,0"
}
```

Then reboot and try `./spotify-connect-web -n Salon -o plug`

redvoodoochild, 2016-11-27:
Thank you very much : It works.
Regards

closed 2016-11-27 by redvoodoochild

jvdgeest, 2017-05-18:
It took me at least an hour to figure out, but you need to edit both `/etc/asound.conf` and `spotify-connect-web-chroot/etc/asound.conf` with the content provided by @tommph :)