Fornoth/spotify-connect-web #96: Spotify connect, control 3rd device

OdysLam, 2016-11-30:
It is possible to use this library to automate spotify control, meaning that i want the device that runs this script(rpi) control another spotify device (pc).

I searched through the docs but I haven't find a clear answer, thank you for your time :)

danifr, 2016-12-06:
Not sure what you mean.
You can only have one active session (same user & pass) open on one of your devices at a time. This is a Spotify restriction.

This also means that you can control the flow of the music from every device you have your account opened on.

Fornoth, 2016-12-12:
This library isn't capable of controlling other spotify devices, you could try https://github.com/badfortrains/spotcontrol though

OdysLam, 2017-03-23:
Actually that's what I am trying to achieve. That library seems a little incomplete but I will give it a try
It's strange that there are a dozen python spotify libraries but none seems to give the spotify connect functionality

Fornoth, 2018-01-15:
You can try Spotify's Web API for controlling other Spotify Connect devices
For example: https://developer.spotify.com/web-api/start-a-users-playback/