Fornoth/spotify-connect-web #97: parallel running of two instances on two Pis

binderth, 2016-12-03:
At present, when I start the chroot-version of spotify-connect-web on two separate Pis, the second Pi is kicked out of the list of Spotify Connect devices in Spotify (App or Desktop). Both work though. But I'd like to have them running both (or up to five - for each family member one for his/her room) at the same time. 

I cloned the SD-Card of the first to the second, so perhaps there is some connection (unique ID?) within spotify-connect-web, which is now the same. 
I do see some "device-id" on startup, which is the same on both Pis. My guess is, to change this id on one of the Pis? But how should I do it?

Thanks for your help!

tommph, 2016-12-04:
The device-id is generated on first run and stored in the credentials file. It will be `credentials.json` in the spotify-connect-web application folder if you haven't specified it. Delete that on one of the Pis and run again and you should get a unique one for that instance,

binderth, 2016-12-04:
Great! Works!

closed 2016-12-04 by binderth