Fornoth/spotify-connect-web #99: Force Rpi to work like tv, computer or tablet instead of Speaker

lollo78, 2016-12-06:
Hi, I'm using the LibreElec addon wrapped from this great project.
All works fine, but my Rpi is recognized like a "Speaker system".
According to Spotify specs, local streaming to Speakers is the only method reserved to premium users: https://support.spotify.com/my-ms/listen_everywhere/supported_connection_types/spotify-connect/
There's a way to set a different device type like Tablet, TV or Computer for the rpi?

Zalastax, 2016-12-07:
I'm sorry but spotify-connect-web uses libspotify which requires Spotify premium to work. [Spotify family](https://www.spotify.com/family/) can be an affordable option if you can share the subscription with some close ones!

Fornoth, 2016-12-12:
What @Zalastax said is correct, this only works for spotify premium accounts

closed 2016-12-12 by Fornoth

lollo78, 2016-12-12:
Hi, are you working on a new version without libspotify?
This libs seems to be unmainteined/deprecated.

Fornoth, 2017-01-02:
I'm not sure if libspotify_embedded is included in that deprecation notice that was posted on the spotify developer site. But I plan on moving to either librespot or whatever the libspotify replacement is (if it has spotify connect support)