Fornoth/spotify-connect-web #101: OverflowError

tomatenparty, 2016-12-11:
After downloading everything and trying to start the programm this error message shows up:

"Loading Spotify library...
Device has no native mute
min_volume_range: 0.0
Using libspotify_embedded version: master-v1.5.0-gfba69d71
SpInit: 0
Traceback (most recent call last):
  File "main.py", line 56, in <module>
    connect_app = Connect(web_error_callback, web_arg_parser)
  File "/usr/src/app/connect.py", line 87, in __init__
    lib.SpPlaybackUpdateVolume(mixer_volume)
OverflowError: integer 112756244 does not fit '16-bit unsigned int'"

I am using the iqaudio dac+. I hope you can help me.

closed 2016-12-11 by tomatenparty

Fornoth, 2016-12-12:
Got it fixed?

tomatenparty, 2016-12-12:
Yes, had to define the mixer, because i was using the iqaudio dac+ hat unit.