Fornoth/spotify-connect-web #103: PCM mixer for HDMI clipping by 4dB in LibreELEC on Raspberry Pi 3

makruiten, 2016-12-27:
On LibreELEC there is no default mixer for HDMI, as Kodi does not use ALSA for the HDMI output. When you install teh Spotify Connect Web add-on, a mixer is added with the name PCM. This mixer has a decibel range with an upper limit at +4dB, which causes the signal to distort due to clipping when you set the Spotify volume at the maximum level. I'm not sure if this is an issue with Spotify Connect Web or with the way it is packaged for LibreELEC, but I think it would be better if the upper limit was set at exactly 0dB by default.

If anything I said is incorrect, or incomplete, please let me know.

awiouy, 2016-12-27:
https://github.com/Fornoth/spotify-connect-web/issues/85

makruiten, 2016-12-27:
Thank you, though I'm not really sure if this is the same problem. For as far as I'm aware I cannot start Spotify Connect Web with arguments on LibreELEC, right? If I can, please tell me how. But I was not really referring to the volume control per se, but the fact that it goes beyond 100%, because the mixer goes to 4dB instead of 0dB. In the past one of the mixers of the HiFiBerry soundcard had this problem, but it's capped at 0dB now in LibreELEC.

awiouy, 2016-12-27:
1. stop the service: systemctl stop service.spotify-connect-web
2. edit/run the service script: /storage/.kodi/addons/service.spotify-connect-web/bin/spotify-connect-web.start

makruiten, 2016-12-27:
The 100% level is still exactly the same as before adding `--dbrange 45`. To be more clear: what the mixer calls 100% is in fact ~200% volume. 100% should be at 0dB. Since the PCM mixer is created by the add-on, I would assume there must be some way to set the volume limit, but I have no idea how and I might be completely wrong. From what I understand, the `dbrange` argument alters the steps between 0% and 100%, but it does not alter the upper limit.

In the past one of the mixers of the HiFiBerry soundcard had this problem (the upper limit being above 0dB), but it's capped at 0dB now for some time.

awiouy, 2016-12-27:
The addon does not create a mixer, it uses existing mixers. There are issues here with respect to erroneous dbrange.