Fornoth/spotify-connect-web #104: spotify_appkey.key generation service not available anymore

danifr, 2016-12-28:
Project's README says:
> Also requires a spotify premium account, and the spotify_appkey.key (the binary version) file needs to be obtained from https://developer.spotify.com/my-account/keys , and needs to placed in the python scripts directory, or have the path specified with the -k parameter

According to Spotify's web, key generation and retrieval service has been deprecated:

> The libspotify key generation and retrieval service has been deprecated, along with libspotify.
You may apply to get a key using our App-key Request Form for private, individual usage. Bear in mind that you should do no new development using libspotify.

Is it possible to make the current version of spotify-connect-web work with the new app-key system?

Anyhow, README should be updated to reflect these changes.
Thanks!!!

plietar, 2016-12-28:
FYI, [librespot](https://github.com/plietar/librespot) works without an app key, so you could use it instead if you are unable to obtain one.

snoupyrelete, 2017-01-15:
Any update on this? I'm trying to get it to work with the magic mirror module for spotify connect. :/

nooryani84, 2017-01-28:
@danifr I only got my key a few weeks ago.   You have to fill out a form and just tell them it's for personal use etc.   It's unfortunate that you have to wait, but at least you can still get it.  Took a few weeks but they did indeed send me a spotify_appkey.key via e-mail

https://developer.spotify.com/technologies/libspotify/application-keys/

PooLP, 2017-03-18:
Hi,
After two request, i have not respond, please, update to use this library without key file !

skraburski, 2017-03-18:
Try again, about month ago I wrote to them with explanation why I need it and next day I had a key.
I've managed to successfully configure with it spotify-connect on my volumio.

Fornoth, 2018-01-15:
It really isn't possible to get this to work without an app key, but since #114 has been merged, the README is updated with the new instructions to get a key

closed 2018-01-15 by Fornoth

ayancey, 2018-09-26:
The URL is 404 again. Why do you need to keep changing shit, Spotify?