Fornoth/spotify-connect-web #105: Multiple login on two devices not possible

berrywhite96, 2016-12-29:
I thought I can use several pis to enjoy spotify everywhere in my house. But I tried to start the application on two pis at the same time, and the playback switched from the one to the other one. In my spotify connect list never appears two speakers, either the first or the second.
There are no error lines in the command line.

Thank you for your support!

berrywhite96, 2016-12-29:
Ok, I found a fix for that. You can't use the same credentials.json more than one time, so just let a new credentials.json generate and use this for the second or third device.

closed 2016-12-29 by berrywhite96

mundodisco8, 2017-04-01:
can you give more detailed instructions about the Avahi config? I've never used it and I have no clue what to do :S

Fornoth, 2018-01-15:
@mundodisco8 I updated the readme with some new instructions, can you try those?