Fornoth/spotify-connect-web #106: Multiple instances on one device

Sindaugh, 2016-12-29:
Hi,

first of all (though I might repeat myself) - I love it. Period.

What I would like to ask though... Can I run multiple (as in more than one) instances at the same time? I have a single RPi connected to my stereo via HifiBerry. While we worked out the blocking issue with Kodi (thanks again) I now would like to run one instance per spotify account. You see sometimes I want to also listen over the stereo while my wife is out running and listening to something else - meaning I can't use her account on the stereo...

So: two accounts on one device - though only one account will be active at a given time.

Thanks a lot.
Steven

berrywhite96, 2017-01-01:
So, I don't see your problem. Do you have a problem with the audio output or what?

Fornoth, 2017-01-02:
You could use zeroconf to allow multiuser https://github.com/Fornoth/spotify-connect-web#passwordlessmultiuser-zeroconfavahi, or just run multiple copies of it at the same time, just don't use the same credentials.json for all of them (see #105)