Fornoth/spotify-connect-web #110: [Raspberry Pi 2] Random Stops in Playback

smartsoldier, 2017-01-19:
I did a fresh install with PiBakery and after my apt-get updates etc I installed spotify-connect-web. I set it up to run at each boot, and it works beautifully...until it doesn't. It will randomly stop playback for 10-30 seconds, and then continue on like nothing happened (maybe skipping that time in the music it is hard to tell). I am thinking it is either Wifi signal or power supply. Am I missing anything software-wise that I could fix easily? If not, how can I test if it is Wifi / power? Thanks.

RafaPolit, 2017-01-24:
Having the same issue on RPi3. There are alsotimes when this behavior is more of a stutter than a complete stop. I believe as well that it is a wifi issue, as during this time any ssh session gets laggy, connecting new sessions to the RPi may timeout, and we servers are also slow to respond.

Haven't thought of power supply issues, but I am using a good iPad charger that are, supposedly, very stable.

There is also not a significant load when this happens, maybe 6% of the CPU and plenty of free memory, so I don't the think it's resources.

Any insight into what it maybe? Thanks,
Rafa.

smartsoldier, 2017-01-26:
Actually I think mine was a power supply issue. I changed outlets and used a different charger and now it works flawlessly.

RafaPolit, 2017-01-28:
That's funny, because I changed from WiFi to wired Ethernet connection and my problems where solved! I'm pretty sure it was a WiFi issue as ssh connections into the RPi where also delayed and copying files took forever.

keab, 2017-02-05:
I've had similar problems with the Spotify client in Volumio (RPi1 hw) . It was caused by the caching of files to SD card not being fast enough. I solved it by caching to a ramdisk instead. A faster SD card might also solve it.  (Strangely enough I don't have the problem after upgrading to Volumio 2)

Fornoth, 2018-01-15:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2018-01-15 by Fornoth