Fornoth/spotify-connect-web #111: stutter in play-mode

binderth, 2017-01-27:
spotify-connect-web works perfect!
My problem is, that I experience some "stuttering" music while in Play. There's nothing big going on in parallel. It's just a PI3 with dedicated spotify-connect-web in chroot-mode. The System idles with not much to do, but it appears, that there's some stutter like a crack in a old Vinyl! ;)
is there some logfile, which I could have a look into, or how do I open up the loglevel of spotify-connect-web for that purpose.

```top - 13:22:44 up 4 days,  1:10,  1 user,  load average: 0.36, 0.33, 0.19
Tasks:  87 total,   1 running,  86 sleeping,   0 stopped,   0 zombie
%Cpu(s): 36.6 us,  1.4 sy,  0.0 ni, 62.0 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem:    445088 total,   132972 used,   312116 free,    12700 buffers
KiB Swap:   102396 total,        0 used,   102396 free.    77868 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
  655 root      20   0   43092  21112   8076 S 33.0  4.7 360:00.28 python
 2931 pi        20   0    5100   2468   2104 R  1.6  0.6   0:02.33 top
 2885 pi        20   0   11472   3944   3300 S  0.6  0.9   0:00.67 sshd
    3 root      20   0       0      0      0 S  0.3  0.0   2:46.78 ksoftirqd/0
    1 root      20   0    5312   3720   2720 S  0.0  0.8   0:21.76 systemd
    2 root      20   0       0      0      0 S  0.0  0.0   0:00.01 kthreadd
    5 root       0 -20       0      0      0 S  0.0  0.0   0:00.00 kworker/0:0H```

PS: It's regardless, if I'm listening to Iron Maiden oder Justin Bieber.

nooryani84, 2017-01-28:
You running it on a stock Raspbian install or something like Volumio?  USB/I2S DAC or the integrated analog output?   I believe I had some stutter when I ran spotify-connect-web on Volumio, though has worked fine with stock Raspbian and RuneAudio (Arch)

binderth, 2017-01-28:
I'm using raw Raspbian with a USB DAC (from CSL). i didn't experience this for the last time, just since let's say two weeks or so? I don't remember updating the Pi or spotify-connect-web since... just curious if someone has same issues. What is RuneAudio - did only find a software called RuneAudio?

nooryani84, 2017-01-28:
You using wireless or wired? RuneAudio is another RaspberryPi distro based on Arch instead of Rasbian.

binderth, 2017-01-28:
I use LAN with my PIs. So, if I'm the only one, I must dig a bit deeper. Perhaps it's the USB DAC, I can use a different one.

nooryani84, 2017-01-28:
The raspberry pi LAN/ethernet shares its bus with the USB so that can cause issues.   I would recommend using wifi, that would probably resolve the issue.

Fornoth, 2018-01-15:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2018-01-15 by Fornoth