Fornoth/spotify-connect-web #113: systemd service

nooryani84, 2017-01-28:
Related to #49

I've been trying to get spotifyconnectweb working seperately on a Raspberry Pi 3 headless installation and had quite a bit of problems creating a systemd service without a seperate script that cd into the directory before running spotifyconnectweb on a new line.   What I've learned with help from looking at other peoples' scripts and trial and error is that unless you create a seperate script like mentioned, it seems to be very picky about accepting arguments such as --username, --password, --key etc.  It just produces an error when attempting to start the service via systemctl start $servicename and doesn't start the service.

It works fine if I use the chrooted version, but not the smaller one with built-in dependencies, which is preferrable to me.

So I solved this by specifying the locations manually, which worked perfectly.

`ExecStart=/pathto/spotifyconnectweb -k /pathto/spotify_appkey.key -n "Spotify Connect" -b 320 -c /pathto/credentials.json`

So on my Raspbian install I have created the following service as `/etc/systemd/system/spotify-connect.service`
```
[Unit]
Description=Spotify Connect
After=network.target
Requires=network-online.target

[Service]
ExecStart=ExecStart=/pathto/spotifyconnectweb -k /pathto/spotify_appkey.key -n "Spotify Connect" -b 320 -c /pathto/credentials.json
Restart=always
RestartSec=10
StartLimitInterval=30
StartLimitBurst=20

[Install]
WantedBy=multi-user.target
```
Just wanted to share this in case it helps anyone else..    Thank you again for your great implementation, my next issue to tackle will be starting avahi on boot.

Would it be recommended to just add this line of code to /etc/rc.local ?
`avahi-publish-service TestConnect _spotify-connect._tcp 4000 VERSION=1.0 CPath=/login/_zeroconf`

Any suggestions or better implementations are welcomed!

jdrossl, 2017-02-14:
This works great! but you should also add the working directory param or the web ui will not work properly.

```
WorkingDirectory=/pathto/
```

nooryani84, 2017-02-14:
Ah, I haven't actually used the web ui.   maybe we should pool our knowledge and add it to the wiki

ashthespy, 2017-03-04:
Tried following your steps also with Rasbian, I have
Created `/etc/systemd/system/spotify-connect.service`
````
[Unit]
Description=Spotify Connect
After=network.target
Requires=network-online.target

[Service]
WorkingDirectory=/home/pi/spotify-connect-web/
ExecStart=/home/pi/spotify-connect-web/spotify-connect-web -k /home/pi/spotify-connect-web/spotify_appkey.key -n "Spotify Connect" -b 320 -c /home/pi/spotify-connect-web/credentials.json -o "sysdefault:CARD=Audio"
Restart=always
RestartSec=10
StartLimitInterval=30
StartLimitBurst=20

[Install]
WantedBy=multi-user.target
````
I did
However when checking the status of the service after starting it, I get
````
● spotify-connect.service - Spotify Connect
   Loaded: loaded (/etc/systemd/system/spotify-connect.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2017-03-04 17:45:37 CET; 3s ago
 Main PID: 1533 (code=exited, status=255)

Mar 04 17:45:37 raspberrypi systemd[1]: spotify-connect.service: main process exited, code=exited, status=255/n/a
Mar 04 17:45:37 raspberrypi systemd[1]: Unit spotify-connect.service entered failed state.
````

I am however able to get it working just fine via ssh by running
````
/home/pi/spotify-connect-web/spotify-connect-web -k /home/pi/spotify-connect-web/spotify_appkey.key -n "Spotify Connect" -b 320 -c /home/pi/spotify-connect-web/credentials.json -o "sysdefault:CARD=Audio"
````

Any ideas?

nooryani84, 2017-03-04:
I'm no expert, but have you tried without

WorkingDirectory=/home/pi/spotify-connect-web/

and without -o "sysdefault:CARD=Audio"

?