Fornoth/spotify-connect-web #118: Volume Contol not working

atze09, 2017-02-03:
Hey guys,
I installed according to documentation on NTC CHIP with usb sound card.

I start with:
./spotify-connect-web.sh --username name --password pass --name chip --device Device

it works playing sound on internal audio jack and on usb sound card, but volume change is not possible.
Can anyone please help me out?

atze09, 2017-02-04:
The output of aplay -L is:
`root@Airplay-CHIP4:~# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
equal
sysdefault:CARD=sun4icodec
    sun4i-codec,
    Default Audio Device
dmix:CARD=sun4icodec,DEV=0
    sun4i-codec,
    Direct sample mixing device
dsnoop:CARD=sun4icodec,DEV=0
    sun4i-codec,
    Direct sample snooping device
hw:CARD=sun4icodec,DEV=0
    sun4i-codec,
    Direct hardware device without any conversions
plughw:CARD=sun4icodec,DEV=0
    sun4i-codec,
    Hardware device with all software conversions
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions
`
./spotify-connect-web.sh --username name --password pass --name chip --playback_device front:CARD=Device --mixer_device_index 1

Is not working propely.
Do i need to change the mixer?

nooryani84, 2017-02-04:
What device are you using?

atze09, 2017-02-04:
I use the usb soundcard "front:CARD=Device" with this line:

./spotify-connect-web.sh --username name --password pass --name chip --playback_device front:CARD=Device --mixer_device_index 1

nooryani84, 2017-02-04:
No I mean Rpi 1,2,3, zero etc

atze09, 2017-02-04:
Sorry.
It is a NTC CHIP, which in my opinion is the ideal device for multiroom/spotify.
I already installed shairport-sync from mike brady, this is working properly with volume control.

atze09, 2017-02-04:
Ca anyone help me in a deeper explanation of those arguments:
 --mixer_device_index MIXER_DEVICE_INDEX
                        alsa card index of the mixer device
  --mixer MIXER, -m MIXER
                        alsa mixer name for volume control

nooryani84, 2017-02-04:
Ah, sorry.   I haven't used one of those, I just know they use Allwinner.   I have only used the chroot and the precompiled with dependencies on the RPi3.   Have you tried both chroot and the other one?

atze09, 2017-02-04:
No i haven't nearly everything is working, exept volume control.
How could chroot help me get it working?

atze09, 2017-02-04:
Now i now what is the root cause, i had to change the following line from.
mixer_volume = int(mixer.getvolume()[0]*655.35)
to
mixer_volume = 1 #int(mixer.getvolume()[0]*655.35)
Because i got teh following error when i start spotify connect.

`Loading Spotify library...
min_volume_range: 0.0
Using libspotify_embedded version: master-v1.5.0-gfba69d71
SpInit: 0
Traceback (most recent call last):
  File "main.py", line 56, in <module>
    connect_app = Connect(web_error_callback, web_arg_parser)
  File "/usr/src/app/connect.py", line 87, in __init__
    lib.SpPlaybackUpdateVolume(mixer_volume)
OverflowError: integer 112756244 does not fit '16-bit unsigned int'
`
How can i do it without changing the volume to 1??

atze09, 2017-02-04:
Saw some fixes should i try any of them?

atze09, 2017-02-05:
With the fix i get the following pylast error:
`Traceback (most recent call last):
  File "loader.py", line 3, in <module>
  File "runpy.py", line 180, in run_module
  File "runpy.py", line 72, in _run_code
  File "/root/spotify-connect-web/main.py", line 15, in <module>
    from connect import Connect
  File "/root/spotify-connect-web/connect.py", line 9, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/root/spotify-connect-web/console_callbacks.py", line 9, in <module>
    from lastfm import lastfm
  File "/root/spotify-connect-web/lastfm.py", line 3, in <module>
    import pylast
ImportError: No module named pylast
`

Can anyone help me?

nooryani84, 2017-02-05:
Are any of these issues present without the USB soundcard?  What soundcard are you using?  Which chipset does it use?

atze09, 2017-02-05:
But this is not spotify Connect or?

Pylast could not be found, is it an issue oft THW allwinner CPU?

nooryani84, 2017-02-05:
I'm not sure, though I believe its better supported on the Raspberry Pi.

atze09, 2017-02-05:
Yes in my pi's it is running smoothly. But the NTC chip is thr cheaper alternative!
Maybe the developer can support me?

nooryani84, 2017-02-05:
I only know that allwinner have not had a good relationship with the open
source community
https://en.wikipedia.org/wiki/Allwinner_Technology#Linux_controversies

You might want to try the CHIP forums as well

Fornoth, 2018-01-15:
Try the latest build
https://github.com/Fornoth/spotify-connect-web/releases/tag/0.0.4-alpha
plus the options `--playback_device 'default:CARD=sun4icodec' --mixer 'Power Amplifier'`. That's what I use on my CHIP.