Fornoth/spotify-connect-web #120: Issue when no default mixer exists

keab, 2017-02-05:
console_callbacks.py requires a default mixer (alsa.mixers()[0]) to exist even if another mixer is explicitly specified with --mixer. 
It's a problem at least in the following case: No HW mixer exists (The case for Hifiberry DAC), and a software mixer has been added to asound.conf and is specified with the --mixer option, but that mixer has never been used by anyone. (It seems - although I don't understand how alsa works - that it becomes visible with amixer as soon as some process has used it once)
The solution now is to edit console_callbacks.py, removing the dependency on alsa.mixers()[0], or to invoke the software mixer once with another program.

This could be solved by skipping the mandatory "default=alsa.mixers()[0])" assignment and only trying that assigment if there is no --mixer argument provided.

Flydroid, 2017-02-27:
It worked for me to add following to the /etc/asound.conf

`pcm.hifiberry {
    type softvol
    slave.pcm "plughw:0"
    control.name "Master"
    control.card 0
}`

and specifying --mixer Master and --playback_device hifiberry when starting spotify connect

keab, 2017-03-03:
Looks similar to mine/volumio's. Did you check with 'amixer' that there was **no mixer at all** in the system before you started?

pcm.softvolume {
    type             plug
    slave.pcm       "softvol"
}

pcm.softvol {
    type            softvol
    slave {
        pcm         "plughw:0,0"
    }
    control {
        name        "SoftMaster"
        card        0
        device      0
    }
max_dB 0.0
min_dB -50.0
resolution 100
}