Fornoth/spotify-connect-web #125: "Radio" playback stops prematurely

aloupos, 2017-02-21:
It seems Spotify changed some things, and now when you are playing radio, the UI shows the first n songs, with a note after that "More songs load as you listen".  The works normally when the device is either local or a native spotify app on another device, however at the last song with the device set to a spotify-connect-web host it doesn't.  

To reproduce, right click a track or artist, and click go to "radio".  Choose the last song in the list (some are grayed but still selectable.  When the list visible song is finished, playback will stop.  The expected behaviour is that new songs load and radio continues indefinitely.  

Great App, thanks for the hard work you put into it!

3onyc, 2017-06-18:
This same issue exists with the daily mixes, I'm not sure whether this is a limitation of libspotify, or something that just needs to be implemented though.

Fornoth, 2018-01-15:
This is most likely because Spotify sends over just a set of songs, and not the actual radio station. Nothing I can fix unfortunately

closed 2018-01-15 by Fornoth