Fornoth/spotify-connect-web #126: Crashing when trying to play local song

ClemaX, 2017-03-23:
Trying to play a local file crashes the latest version on my raspberry pi. I know that local files can't be played over Spotify Connect but when I accidentally select one it crashes and I need to restart spotify-connect-web manually.

Here is what I can see being logged:
```
playback_seek: 0
kSpPlaybackNotifyTrackChanged
kSpPlaybackNotifyBecameActive
Segmentation fault (core dumped)
```

joeltimothyoh, 2017-11-02:
I think this might be connected to the fact that SCW crashes that happen when attempting to play an unavailable track on Spotify. Seems like the only workaround is to play only tracks that are playable on Spotify.

Fornoth, 2018-01-15:
This really isn't something I can fix unfortunately, since I don't control the actual playback. Although I'll look into it to see what `SpCallbackPlaybackUnavailableTrack` is.