Fornoth/spotify-connect-web #130: Launching any application stops playback

AragurDEV, 2017-06-05:
Song is playing on my RaspberryPi.
If I start my Android App and my Windows App isn't opened played track will stop immediately.
Than it almost isn't possible to start a song agian via my RaspberryPi.
If I select the device 2 seconds later it get's unselected. Also this happens a lot if I want to change the track.
I have to click on the song and immediately exit the app.
If I start the Windows app. RaspberryPi playback also get's stopped. But casting a song works fine.

Using 0.0.3 of spotify web connect.
Newest Android Version (newest Beta because Official wasn't playing any song at all on my RaspberryPi)
Latest Version of Spotify on Windows.

kunterbunth, 2017-07-04:
Try to install with docker. The installer build is very old.

AragurDEV, 2017-07-05:
@kunterbunth i tried installing it with docker
But it keeps failing at step 8
Command output: https://pastebin.com/zwvYAG0A

kunterbunth, 2017-07-05:
Would you look at my pull request? I updated the docker file because I got the same error messages. A dependency was/is missing.

AragurDEV, 2017-07-05:
okay thanks now it works
But how to start it?
~/run-with-docker is not working. returns no such file or directory

AragurDEV, 2017-07-06:
@kunterbunth any assistance?

_Sent from my LGE Nexus 5X using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

kunterbunth, 2017-07-06:
You need to locate the run-with-docker file and execute it from there. ~/ just means run xyz from the home directory. Should be more like ~/spotify-connect-whatever/run-with-docker if you put the master directory from github in your home directory

tenortim, 2017-08-10:
I'm experiencing similar issues. Everything worked fine until fairly recently. Now if I start playing from the iPhone Spotify app, switch to another app, and switch back, playback stops and Spotify moves the playback device to my iPhone (I see a kSpPlaybackNotifyBecameInactive has been sent to spotify-connect-web). This doesn't happen if I'm e.g. connected to the laptop or some other device. In fact, at this point, even clicking on the device at the bottom of the screen immediately stops playback.

I have to wonder if the libspotify_embedded_shared.so library we're using is interacting correctly with the Spotify servers or if there's some incompatibility (since they aren't updating it but haven't replaced it either).