Fornoth/spotify-connect-web #131: No sound with Raspbian Stretch Lite and Hifi-Berry clone

ykelle, 2017-08-22:
I was trying to install it from source on an Raspberry Pi 2 with Raspbian Stretch Lite fresh install. I am using a Hifi-Berry clone (Pifi DAC+). It seems like to work after I start the script with
`LD_LIBRARY_PATH=$PWD python main.py --bitrate 320 --name Test --playback_device hw:CARD=sndrpihifiberry --mixer Digital`
However when I start a song there is no sound. I then tested the sound card with mplayer which shows the sound card works.

I have redone the exact same installation with a Raspbian Jessie Lite image. With Jessie Lite all works fine. I also tested the onboard sound card which works perfectly with the script.

Also tested the `--mixer_device_index` parameter.
`LD_LIBRARY_PATH=$PWD python main.py --bitrate 320 --name Test --playback_device hw:CARD=sndrpihifiberry --mixer_device_index 0`
No success

`aplay -L` output:
null
    Discard all samples (playback) or generate zero samples (capture)
default
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
sysdefault:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dacplus, 
    Default Audio Device
dmix:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus, 
    Direct sample mixing device
dsnoop:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus, 
    Direct sample snooping device
hw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus, 
    Direct hardware device without any conversions
plughw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus, 
    Hardware device with all software conversions

celeryclub, 2017-09-26:
Also experiencing this issue. Relevant logs (notably "Unable to acquire device"). Any help would be appreciated! Thanks for building this software.

`ESDK: callback context (nil), event 15, source: (nil), param (nil)
ESDK: callback context (nil), event 1001, source: (nil), param 0x76100670
ESDK: callback context (nil), event 8, source: (nil), param (nil)
ESDK: Calling SpCallbackPlaybackSeek
playback_seek: 105882
Requesting Bytes: 4138k => 4266k +128k (used: 0k, buf: 128k, thres: 48k, rate: 320k)
ESDK: Notifying kSpPlaybackEventAudioFlush
kSpPlaybackEventAudioFlush
ESDK: Notifying kSpPlaybackNotifyPlay
kSpPlaybackNotifyPlay
Unable to acquire device:  Device or resource busy [hw:1]
WARNING: Underrun in download buffer!
WARNING: Underrun in download buffer!
WARNING: Underrun in download buffer!`

joeltimothyoh, 2017-09-30:
Yes, spotify-connect-web stopped working on Stretch.

Perhaps related #133

mickvdv, 2017-10-29:
I had similar issues. I created some code that replaces the usage of alsaaudio with pymedia. Which fixes the issues I had. I guess it uses the default audio output, so if you have configured that correctly it should work.

See: https://github.com/Fornoth/spotify-connect-web/pull/136

ytzelf, 2018-02-22:
hi, any news on this? Thnaks

vdbee, 2018-02-24:
I had similar issues. If I tried to route the sound from spotify-connect-web via dmix to my i2s-soundcard it did not work. But when I directly routed the sound to my i2s-card it worked. However, with this routing, I cannot combine the usage of the soundcard with OSMC. hence, I reverted back to Jessie for now. Maybe it is more a Jessie-Alsa problem?

vdbee, 2018-06-16:
Any news here? I just tried again with all updates at Stretch but it still doesn't work via dmix. I revert back to my images for Jessie