Fornoth/spotify-connect-web #133: Doesn't work anymore?

ZatarraNL, 2017-09-08:
After years of satisfied use, it seems this app no longer works. Can anyone confirm?

worksasintended, 2017-09-09:
Still works for me, also the newest version.

joeltimothyoh, 2017-09-30:
I can confirm that spotify-connect-web works on Raspbian Jessie, but not on Raspbian Stretch. I get the exact issues as said in issue #67. I can get spotify-connect-web to run by adding asound.conf with default settings, but there will be no sound, instead getting the following errors:

```
ALSA lib confmisc. c: 674: (snd determine driver) could not open control for card O 
ALSA lib conf. c: 3328: (snd config hooks call) function snd config hC"3k load for a 
11 cards returned error: Invalid argument 
ALSA lib conf . c: 4687: (snd config expand) Unknown parameters CARL—ALSA 
ALSA lib pcm. c: 2217: (snd pcm open noupdate) Unknown PCM sysdefault : CARD—ALSA 
Unable to acquire device : Invalid argument [sysdefault : CARD=ALSA]
```

I spent quite some time trying to work around the issue, but to no avail, and have resorted to running Raspbian Jessie for now.

worksasintended, 2017-10-02:
Now I got problems too, also in the dockered version: 
Python errors during docker built: 
`Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 22, in <module>
    import requests, six
  File "/usr/lib/python2.7/dist-packages/requests/__init__.py", line 68, in <module>
    _attach_namespace(urllib3, 'requests.packages')
  File "/usr/lib/python2.7/dist-packages/requests/__init__.py", line 63, in _attach_namespace
    module = __import__(name)
  File "/usr/lib/python2.7/dist-packages/urllib3/contrib/pyopenssl.py", line 55, in <module>
    import OpenSSL.SSL
  File "/usr/lib/python2.7/dist-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import rand, crypto, SSL
  File "/usr/lib/python2.7/dist-packages/OpenSSL/rand.py", line 11, in <module>
    from OpenSSL._util import (
  File "/usr/lib/python2.7/dist-packages/OpenSSL/_util.py", line 4, in <module>
    binding = Binding()
  File "/usr/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py", line 89, in __init__
    self._ensure_ffi_initialized()
  File "/usr/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py", line 113, in _ensure_ffi_initialized
    libraries=libraries,
  File "/usr/lib/python2.7/dist-packages/cryptography/hazmat/bindings/utils.py", line 80, in build_ffi
    extra_link_args=extra_link_args,
  File "/usr/lib/python2.7/dist-packages/cffi/api.py", line 340, in verify
    lib = self.verifier.load_library()
  File "/usr/lib/python2.7/dist-packages/cffi/verifier.py", line 75, in load_library
    return self._load_library()
  File "/usr/lib/python2.7/dist-packages/cffi/verifier.py", line 151, in _load_library
    return self._vengine.load_library()
  File "/usr/lib/python2.7/dist-packages/cffi/vengine_cpy.py", line 149, in load_library
    raise ffiplatform.VerificationError(error)
cffi.ffiplatform.VerificationError: importing '/usr/lib/python2.7/dist-packages/cryptography/_Cryptography_cffi_813c10e0x7adb75f8.arm-linux-gnueabi.so': /usr/lib/python2.7/dist-packages/cryptography/_Cryptography_cffi_813c10e0x7adb75f8.arm-linux-gnueabi.so: symbol SSLv2_client_method, version OPENSSL_1.0.0 not defined in file libssl.so.1.0.0 with link time reference`
While my locally installed version (which worked close to two years returns:
`
  File "main.py", line 27, in <module>
    connect_app = Connect(web_error_callback)
  File "/usr/src/app/connect.py", line 42, in __init__
    in json.loads(f.read()).iteritems() })
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded`