Fornoth/spotify-connect-web #138: Vanilla Rpi3 + Raspbian issue

banzai200, 2017-11-27:
When i execute the command to open spotify-connect-web, i get the following error:

Loading Spotify library...
Traceback (most recent call last):
  File "loader.py", line 3, in <module>
  File "runpy.py", line 180, in run_module
  File "runpy.py", line 72, in _run_code
  File "/home/pi/spotify-connect-web/main.py", line 14, in <module>
    from connect import Connect
  File "/home/pi/spotify-connect-web/connect.py", line 9, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/home/pi/spotify-connect-web/console_callbacks.py", line 24, in <module>
    audio_arg_parser.add_argument('--mixer', '-m', help='alsa mixer name for volume control', default=alsa.mixers()[0])
alsaaudio.ALSAAudioError: Invalid argument [default]
Failed to execute script loader

I'm using the "-o hw:0" command as instructed in similar issues, but no luck, any help on the issue?

MrJoki007, 2018-01-13:
Same problem with RPi3 on OSMC

Fornoth, 2018-01-15:
Try using the name from the output of `aplay -L`
For example, I get
```
chip@chip:~$ aplay -L
default:CARD=sun4icodec
    sun4i-codec, 
    Default Audio Device
```
So I used `--playback_device 'default:CARD=sun4icodec'`

MrJoki007, 2018-01-16:
Im using the HDMI Output so what should i pick?

root@osmc:~# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions

Fornoth, 2018-01-23:
I'm not sure exactly, but I would try either `default:CARD=ALSA` or one of the `hw:*` or `plughw:*` devices

banzai200, 2018-01-26:
I've solved the issue actually, I'm just not sure how i did

But if i recall correctly, it had something to do with config.txt, because it did work with hdmi but not the aux port

As soon as i get an hold of my pi, I'll post my results here

ytzelf, 2018-02-22:
any news on this? Thanks

st0nec0ld, 2018-05-12:
Same her, any news?

fresh install "2018-03-13-raspbian-stretch-lite", nothing else.

```
pi@raspberrypi:~/spotify-connect-web $ ./spotify-connect-web
Loading Spotify library...
Last.fm: incomplete credentials, not launched
Traceback (most recent call last):
  File "loader.py", line 3, in <module>
  File "runpy.py", line 180, in run_module
  File "runpy.py", line 72, in _run_code
  File "/home/pi/spotify-connect-web/main.py", line 15, in <module>
    from connect import Connect
  File "/home/pi/spotify-connect-web/connect.py", line 9, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/home/pi/spotify-connect-web/console_callbacks.py", line 25, in <module>
    audio_arg_parser.add_argument('--mixer', '-m', help='alsa mixer name for volume control', default=alsa.mixers()[0])
alsaaudio.ALSAAudioError: Invalid argument [default]
Failed to execute script loader

```

TomAvecLeVdsl, 2018-10-21:
Same error can you solved it ?

closed 2020-09-22 by banzai200