Fornoth/spotify-connect-web #139: Volume control is controlling different mixer

thoughtgap, 2018-01-15:
Hello,

first of all thanks for the new release 0.0.4 to which I have updated today on my Rpi3 today running the latest Raspbian.

I cannot work out how to configure the alsa mixer settings for volume control to work correctly in this or earlier versions.

There's a USB sound card connected to the pi.

## `aplay -l`
```
pi@raspberrypi:~/spotify-connect/spotify-connect-web $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
````

## `/proc/asound/cards`
```
pi@raspberrypi:~/spotify-connect/spotify-connect-web $ cat /proc/asound/cards
 0 [ALSA           ]: bcm2835 - bcm2835 ALSA
                      bcm2835 ALSA
 1 [Device         ]: USB-Audio - USB Audio Device
                      C-Media Electronics Inc. USB Audio Device at usb-3f980000.usb-1.2, full speed
````

## `/etc/asound.conf`
```
pi@raspberrypi:~/spotify-connect/spotify-connect-web $ cat /etc/asound.conf
pcm.!default {
type hw
card 1
}

ctl.!default {
type hw
card 1
}
```

## Spotify-connect-web
I run spotify-connect-web like this:
````
pi@raspberrypi:~/spotify-connect/spotify-connect-web $ ./spotify-connect-web --bitrate 320 
--name pi
--playback_device hw:1
````

When I change the volume in Spotify, the "logs" show the correct settings:

````
kSpPlaybackNotifyPlay
AlsaSink: device acquired
playback_volume: 9174
corected_playback_volume: 13
playback_volume: 34733
corected_playback_volume: 52
playback_volume: 56360
corected_playback_volume: 85
````

When I run `alsamixer` I can see that this volume I set in Spotify arrives in the default mixer which looks like this. When I change the volume in Spotify this mixer volume gets changed, too.

[image]

When I switch devices via F6 I can select the USB sound card:

[image]

And manually change the volume there:

[image]

It's the same if I run spotify-connect-web with `--mixer_device_index 0`.  Any other Mixer Device Index will lead to this error:

```ssh
pi@raspberrypi:~/spotify-connect/spotify-connect-web $ ./spotify-connect-web --bitrate 320 --name pi --playback_device hw:1 --mixer_device_index 1
Loading Spotify library...
Last.fm: incomplete credentials, not launched
Traceback (most recent call last):
  File "loader.py", line 3, in <module>
  File "runpy.py", line 180, in run_module
  File "runpy.py", line 72, in _run_code
  File "/home/pi/spotify-connect/spotify-connect-web/main.py", line 15, in <module>
    from connect import Connect
  File "/home/pi/spotify-connect/spotify-connect-web/connect.py", line 9, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/home/pi/spotify-connect/spotify-connect-web/console_callbacks.py", line 105, in <module>
    mixer = alsa.Mixer(args.mixer, **mixer_card_arg)
alsaaudio.ALSAAudioError: Unable to find mixer control PCM,0 [hw:1]
Failed to execute script loader
```

How can I get the default mixer to control the USB sound card?

Any help is greatly appreciated. Thanks in advance!

fertigfutter, 2018-02-18:
The way i did it for my allo Boss Raspberry DAC hat:

- "aplay -L" gave me the name of my device: 'BossDAC'
- "amixer" gave me the name and device index of the wanted control: 'Digital' and '0'

This led to this command line:
"spotify-connect-web [...] --playback_device hw:BossDAC --mixer 'Digital' --mixer_device_index 0"

The same for my LH Labs Geek Out USB device:

- "aplay -L" gave me the name of my device: 'G1V5'
- "amixer" gave me the name and device index of the wanted control: 'Geek Out HD Audio 1V5 ' and '1'

This led to this command line:
"spotify-connect-web [...] --playback_device hw:G1V5 --mixer 'Geek Out HD Audio 1V5 ' --mixer_device_index 1"

Good luck

thoughtgap, 2018-02-18:
Thank you for your help, it's finally working :-)

For reference:

#`aplay -L`
```
pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
equal
plugequal
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions
```

#`amixer` (the second is the one I want)
```
pi@raspberrypi:~ $ amixer
Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback -8109 [20%] [-81.09dB] [on]
pi@raspberrypi:~ $ amixer -c 1
Simple mixer control 'Speaker',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 37
  Mono:
  Front Left: Playback 19 [51%] [-18.00dB] [on]
  Front Right: Playback 19 [51%] [-18.00dB] [on]
Simple mixer control 'Mic',0
  Capabilities: pvolume pvolume-joined cvolume cvolume-joined pswitch pswitch-joined cswitch cswitch-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: Playback 0 - 31 Capture 0 - 35
  Mono: Playback 16 [52%] [-7.00dB] [off] Capture 20 [57%] [8.00dB] [on]
Simple mixer control 'Auto Gain Control',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
```

Volume control does work now with the command:

```
spotify-connect-web --playback_device 'front:CARD=Device,DEV=0'
--mixer_device_index 1
--mixer Speaker
````

closed 2018-02-18 by thoughtgap