Fornoth/spotify-connect-web #146: Link for API key is 404

daniellhotka, 2018-06-20:
I tried to follow the link in the readme:
[https://developer.spotify.com/technologies/libspotify/application-keys/](url)
But got a 404 error.

I proceeded to the new developer console and created a new app, and it gave me  Client ID and Client Secret codes. Is it possible to take these and craft them into a `spotify_appkey.key`? If so, what format would it have to be in?

Thanks

FranzSw, 2018-06-29:
I'm having this issue too...

Latek, 2018-08-30:
Is there any way to solve this issue? Should I use Client ID or Client Secret codes somehow to create spotify_appkey.key? Please help!

crimsonclyde, 2018-09-15:
Same prob.

I've searched around to find a new way to getting the key file but hopeless.

jebos, 2018-09-17:
tbh, i think it's dead. (please correct me if i'm wrong on this). The point is, as far as I understand this entire topic now, the dependency libspotify_embedded_shared.so is not provided by spotify anymore and is deprecated. The lib started to disfunction months ago, though it still can be used if you have it. 

The reality is, spotify wants you to use the web api (for what you need the client id and secret.) Checkout 
python spotipy (link on spotify developer web). 

So technically what you get with those is a link where the user can manually authorize your app. Your app receives a temporary valid token and a refresh token (which can be used by the app to receive a new temporary token). With that token playing a song on a specific detected device is possible.

My conclusion on this is, all those spotify solutions based on libspotify_embedded_shared.so are basically deprecated an will break sooner or later. :(

Wqrld, 2019-02-28:
Has anyone found any alternatives? I really loved the ability to just stream on my dumb speakers without having to buy a overpriced set.

FranzSw, 2019-02-28:
@Wqrld Yes, check out Moodle Audioplayer.

It works without API Key and has a lot more to offer.

Wqrld, 2019-02-28:
> @Wqrld Yes, check out Moode Audioplayer.
> 
> It works without API Key and has a lot more to offer.

Looks nice, sad to see its a full-on image though and not something you can just install on the raspberry with all my projects.

ThomasSchlien, 2019-02-28:
For your own projects I think [librespot](https://github.com/librespot-org/librespot) is the way to go.

Wqrld, 2019-02-28:
> For your own projects I think [librespot](https://github.com/librespot-org/librespot) is the way to go.

i believe cargo wasn't working on the pi but ill try that. I won't respond here anymore as this is a old issue and i dont want to be emailing all of you :)

fertigfutter, 2019-03-01:
Easiest way would be "spotifyd" ([https://github.com/Spotifyd/spotifyd](url)) i think. Key is optional. Ready to use binaries available.

Wqrld, 2019-03-01:
Seems like I have no choice than to tinker with cargo again, moose also looks good and has not support so i might just get a additional pi for music and use that.

fertigfutter, 2019-03-01:
What's "moose"?
No choice? What's wrong with spotifyd? No "build" needed. One simple executable for Pi (Armv6): https://github.com/Spotifyd/spotifyd/releases/tag/v0.2.5
There IS Spotify support in Moode Audio Player.

Wqrld, 2019-03-01:
Autocorrect and my bad about the build thing, thanks for your help

rootkiwi, 2019-08-13:
For Raspberry Pi `raspotify` (https://github.com/dtcooper/raspotify) could also be used. Which uses librespot.